retr0rangepi/RetrOrangePi #1: [Request] Orange Pi Prime support

hutzero, 2017-09-28:
Orange Pi Prime is released for 5 months but still don't have RetrOrangePi.

retr0rangepi, 2017-09-29:
Actually not a RetrOrangePI issue.
OrangePi Prime still lacks GPU Mali drivers, what we don't develop.

closed 2017-09-29 by retr0rangepi

GoogleBot42, 2018-01-20:
Sorry to post on something closed... If i understand correctly, there is now support for h5's Mali 450. https://free-electrons.com/blog/mali-opengl-support-on-allwinner-platforms-with-mainline-linux/

H3 and h5 seem to have about the same level of support in mainline Linux. http://linux-sunxi.org/Linux_mainlining_effort

The primary issue i see is that h5 is 64 bit so there will have to be a separate build for it.

alerinoreis, 2018-01-22:
it's promising, but still preliminary. There are other complex steps that will need more time to get it running.