retr0rangepi/RetrOrangePi #5: Changing overscan on OrangePi Zero

nixenos, 2017-11-04:
When I want to change resolution on my install it is always breaking up my whole system, showing only "A start job is running for Remount Root and Kernel ../no limit"
So I suppose every change in script.bin causes that, but suprisingly when I use tool from yours repo it works, but only for default resolution ( I need less in X dimension so it doesn't help much)

retr0rangepi, 2017-11-04:
After changing resolution using our tool, edit script.bin again manually, reducing width and height always in multiples of 8 , from 672 to 664 , and from 448 to 440. It's likely to work.

nixenos, 2017-11-04:
Okay, I'm gonna try and give feedback. Thanks for response!

nixenos, 2017-11-05:
It works! Thanks a lot!

closed 2017-11-05 by nixenos