retr0rangepi/RetrOrangePi #6: No video in kodi

nixenos, 2017-11-05:
Yet another trouble on my OrangePI Zero. When I want to run a movie from pendrive or from yt it only plays sound, no video or controls at all.

retr0rangepi, 2017-11-05:
right click on video and choose Play with Internal Player, not MPV , which needs a special parameter to run on non-HD resolutions, will work on a fix to work with MPV (hardware accelerated)

nixenos, 2017-11-05:
Will try it in a second. Is there a way to do it using Xbox 360 pad?

retr0rangepi, 2017-11-05:
yes, one of the buttons should be right-click

nixenos, 2017-11-05:
Okay, I figured it out how to do it. 
Now it works, thanks! Unfortunately 720p is too much for opi zero, it shutters as hell but works!

retr0rangepi, 2017-11-05:
i'm reopening so i can add an out-of the-box solution

nixenos, 2017-11-05:
Okay
Well I wonder if there's a trick or anything that would allow me to watch at least 480p videos shutterless, but I think there's no way, Mali is not powerful enough

retr0rangepi, 2017-11-05:
wait for a fix to use MPV instead. Strange that i watch IPTV with internal player without any stuttering

retr0rangepi, 2018-01-10:
edit ~/.kodi/userdata/playercorefactory.xml and replace --vo=vdpau by --vo=x11

closed 2018-01-10 by retr0rangepi