retr0rangepi/RetrOrangePi #7: Splashscreen doesn´t work: solution found

fernandofei, 2017-11-06:
When I try to change the splashscreen in a existing Retropie image, the player can not show the video. 
I checked the script /home/pi/RetroPie-Setup/scriptmodules/supplementary/splashscreen.sh

And at line 240, tries to execute "mpv", but by default, is not installed. You can install 
#sudo apt-get install mpv

If the following message appears, you can just send the command:
#sudo apt-get install -f

	pi@retropie:/home/pi/RetroPie/splashscreens# sudo apt-get install mpv
	Reading package lists... Done
	Building dependency tree
	Reading state information... Done
	You might want to run 'apt-get -f install' to correct these:
	The following packages have unmet dependencies:
	 libsdl2-dbg : Depends: libsdl2-2.0-0 (= 2.0.5+5mali) but 2.0.6+1mali is to be installed
	 mpv : Depends: libavdevice55 (>= 6:11~beta1) but it is not going to be installed
		   Depends: libavfilter5 (>= 6:11~beta1) but it is not going to be installed
		   Depends: libegl1-mesa (>= 7.8.1) but it is not going to be installed or
					libegl1-x11
		   Depends: libguess1 but it is not going to be installed
		   Depends: libva-glx1 (>= 1.0.3) but it is not going to be installed
		   Depends: libva-x11-1 (>= 1.0.3) but it is not going to be installed
		   Depends: libwayland-egl1-mesa (>= 10.0.2) but it is not going to be installed or
					libwayland-egl1
	 ppsspp-odroid : Depends: libsnappy1 but it is not going to be installed
					 Depends: libzip2 (>= 0.10) but it is not going to be installed
	E: Unmet dependencies. Try 'apt-get -f install' with no packages (or specify a solution).

And try again:
#sudo apt-get install mpv

After the install, is possible to execute the video.

Now you can just restart your system and the splashscreen with video will work fine :-)

retr0rangepi, 2017-11-06:
by default , MPV is installed in all images, except one, OrangePi Zero slim image. We can consider it's gonna become a bug since we're adding slim images to all boards soon.

harryzimm, 2017-11-06:
Noted.

fernandofei, 2017-11-06:
I have a orangepi pc, the older version of all hehehe.
I can install in another card to try to reproduce the error and check if this happen with my image or with a fresh new installation to.

Best regards

retr0rangepi, 2017-11-06:
on a closer look, you probably tried to enable splashscreen in an old retropie build. A lot has changed, RetrOrangePi is based on Armbian for at least a year and includes mpv by default.

retr0rangepi, 2017-11-06:
Flash a RetrOrangePi v3.0.1 image, update to v4 and help troubleshooting . Anyway, a bug was discovered before it was born, :p

fernandofei, 2017-11-06:
:-)

retr0rangepi, 2018-07-24:
Added to slim v4.2.

closed 2018-07-24 by retr0rangepi