retr0rangepi/RetrOrangePi #11: Tried installing to eMMC, fails to boot

ArtemKuchin, 2017-12-31:
Latest 4.1
OrangePi plus 2E
Run ROPI 4.1, chose to install into emmc, it showd progress to 100%, asked to remove sd card
and reboot.
I reboot and now it does not boot at all from emms. Complains that emmc0 or something not found etc (scrolling too fast)
Boots ok from sd card.

alerinoreis, 2017-12-31:
Emmc install is still broken on 4.1
Will try to fix it soon

jsmolina, 2018-01-12:
I had the same issue 

I tried installing from RetroPie-setup forked scripts in ARMBIAN from emmc, but that's really a pain. I've been able to compile everything and at I end with SDL not finding video device.

x-manx, 2018-01-26:
To all of you, there are 2 new script files: "emmc_install_v4.sh" and "emmc.instalv4.SH"
Press START and choose first option Desktop
insert a pendrive with these 2 files
Copy the first script into /home/pi/RetroPie/retropimenu and
the second into /home/pi/RetroOrangePi
reboot and choose "emmc_install_v4.sh" in the retropie menu.

alerinoreis, 2018-01-26:
Not needed at all. Just select the option Update RetrOrangePi and the new script will appear. Works just fine.

closed 2018-01-26 by alerinoreis

x-manx, 2018-01-26:
Ok.