retr0rangepi/RetrOrangePi #14: Posible Bug with video splash in RetroOrangePi 4.1 for Pi PC

Klausus, 2018-01-29:
Hi.
I have found a fault in your images for OPI PC.
The images of the version 4.1 for the OPI PC do not allow to use videos like splash.
The mp4 video that comes by default (or any video) does not go. It only plays the audio.

I have tried with 2 different SD.
I have tested with 2 different OPI PCs.
I have tested the image of the USA server and the European server.

Just installed the image, loads retro orange pi, go to menu of options, open retropie-setup.
Configuration Tools -> 831 Splashcreen.
Splashscreen enabled.
Randomizer disabled.
Choose Splashscreen -> own / extra -> splash-720-only.mp4
back, back, reboot ...

When you restart, you only hear the splashscreen audio, there is no video.

Thanks.

retr0rangepi, 2018-01-29:
i will try to download an image from the website again. 
For now, run 'sudo nano /opt/retropie/supplementary/splashscreen/asplashscreen.sh'
change the line beginning with mpv to: 
LD_LIBRARY_PATH=/usr/lib/GLSHIM mpv -vo sdl -fs "$line"

Klausus, 2018-01-29:
I try a few minutes.
Not work.

Imagen loads OK.
Vídeos like splash-720-only.mp4 not load.
Black screen and sound of the video :/

retr0rangepi, 2018-01-29:
Press F4 to quit Emulationstation, then run from command line: 
LD_LIBRARY_PATH=/usr/lib/GLSHIM mpv -vo sdl -fs ~/RetroPie/splashscreens/splash-720-only.mp4
upload a screenshot if you can

Klausus, 2018-01-29:
SDL_CreateWindow failed
No supported renderer

Images: https://imgur.com/a/LAkyI

Klausus, 2018-01-29:
I tried also conect the Pi to internet and update all.

RetroOrange update script.

Retropie script update.

Sudo apt-get update 
Sudo apt-get upgrade

And not works :(

This is the stock image from http://www.retrorangepi.org (for Pi PI tried also conect the Pi to internet and update all.

RetroOrange update script.

Retropie script update.

Sudo apt-get update 
Sudo apt-get upgrade

And not works :(

This is the image stock image from http://www.retrorangepi.org for Pi PC.

Is posible that this is not the last stable version? If this is the problem can you update the image?

If not is this the problem, and the upload image is ok... Why not work?

Thanks 4 All.

retr0rangepi, 2018-07-24:
fixed in v4.2,  release soon

closed 2018-07-24 by retr0rangepi