retr0rangepi/RetrOrangePi #15: OrangePi Zero Plus 2 H5 not working

vgoffman, 2018-02-09:
I have an issue with OrangePi Zero Plus 2 H5. I cant install RetrOrange. There is no any H5 images. H3 only. Can you help me to install manually RetrOrange with Kodi only on Zero Plus 2 H5? Thank you

retr0rangepi, 2018-02-09:
From our website http://www.retrorangepi.org
RetrOrangePi is a non profit gaming and media center distribution compatible with boards equipped with CPU H3 and GPU Mali 400 (most Orange Pis, Banana Pi M2+, Beelink X2 and NanoPi M1).
Until we get working GPU drivers, there will be no support for new boards like Orange Pi PC2, Orange Pi Win, Orange Pi Prime or any other with Mali 450 GPU or A64/H5 CPU.

closed 2018-02-09 by retr0rangepi

vgoffman, 2018-02-09:
When it will be support?

retr0rangepi, 2018-02-09:
Unsure. It's been more than a year of release and the board manufacturer still havent sorted it out. There is work in progress, so keep watching the scene.