retr0rangepi/RetrOrangePi #16: Not really an issue

PTDreamer, 2018-02-15:
Not really an issue, but I was unsure about how to contact the team.
I have read on several place that the nano pi m1 is supported however an can't find any downloadable image. 
Is it still WIP or is it available somewhere?
Thank you

retr0rangepi, 2018-02-16:
Still WIP. We've been running out of spare time lately so a couple of images are still missing, but they'll be available sometime soon. Anyway, i think it's **very likely** to work with a Banana Pi M2+ image. Flash to sdcard, then replace script.bin (in the FAT partition) by the one for NanoPi M1 Plus located in the bin folder (you can remove the original then paste the new file as script.bin)

PTDreamer, 2018-02-19:
Thanks, it worked, I'm having a few instability issues which seem to be caused by wrong dram frequency setting, 672MHz vs 624MHz. Seems like the fex setting is not being used. That is another issue though so closing this. 
Thanks

closed 2018-02-19 by PTDreamer