retr0rangepi/RetrOrangePi #19: Web Interface is not working in Kodi during playback video

BondarenkoArtur, 2018-03-04:
It seems that mpv (or what else used for hardware acceleration) is broke remote web interface (I'm talking about [this](https://kodi.wiki/view/Web_interface) one).
It works fine in all menus and settings but completely freezes when you starting to play any movie. Shows duration of movie 00:01 and ignores any inputs so I cannot stop playback or turn off OPi.

It was tested on Orange Pi One, if any additional information needed, please add a comment.
Thanks in advance.

BondarenkoArtur, 2018-03-04:
P.S. it seems that I can use old version of kodi (probably 16), I would be thankful if somebody will explain me how to set it by default for booting.

retr0rangepi, 2018-03-04:
Kodi 16 (OpenELEC) can be set to default by selecting it from EmulationStation main menu. To return to ES from OpenELEC, click on Programs - Retrorangepi. Builtin video player in Kodi 17 is not hardware accelerated, but has acceptable performance < 720p

BondarenkoArtur, 2018-03-05:
Is it possible to do somehow by ssh ? 
I found few scripts that named boot_to_kodi.sh and others, how is possible to run Kodi 16 (OpenELEC)? where is it located and what I need to change in script?

retr0rangepi, 2018-03-05:
Yes, it's possible. The OpenELEC option in EmulationStation just runs the following command:
'sudo mkimage -C none -A arm -T script -d /boot/boot.kodi.cmd /boot/boot.scr' 
then reboot

BondarenkoArtur, 2018-03-06:
Seems it works, Thanks
Is it possible to somehow enable ssh by changing file on SD card? (in OpenElec)

retr0rangepi, 2018-03-06:
Can't you just enable it from the settings menu?

BondarenkoArtur, 2018-03-06:
Currently no, because I don't have any HDMI TV :(

But anyway this issue with web interface is still appearing on v17 of Kodi.
If it's possible i think it would be great to fix it.
If not, issue might be closed.

retr0rangepi, 2018-03-06:
i need to understand first how the web interface works, i will try something tonight

retr0rangepi, 2018-03-08:
it works fine with internal VideoPlayer, but as said, it's not hardware accelerated.