retr0rangepi/RetrOrangePi #20: The eMMC isn't visible as block device on Orange Pi Plus 2e

lictw, 2018-03-04:
Hello! 

I don't see the eMMC in /dev/mmc* on my Orange Pi Plus 2e. There is only the SD card and its partitions. But the eMMC is visible on the clear armbian. Is it my local problem or an OS' bug? Thanks!

retr0rangepi, 2018-03-05:
probably both, depending on the state of the OS, eMMC is detected. In this forum thread, there's some useful, maybe a fix for you. Please test and let us know.

lictw, 2018-03-06:
Hi!

When i copied the entire /boot from the lateast armbian (all files with override, from clean ISO, not from my eMMC installation) to /boot section of ROPi, exlude zImage and uInitrd files, my board starts to load into the armbian from eMMC. 

I think i do something wrong.. It's hard for me to understand what i'm doing, since this is my first single-board computer. And the boot process is very different from classic PCs.

lictw, 2018-03-09:
Well, i'll try and write here about the results.

retr0rangepi, 2018-08-01:
this is supposed to be fixed in the new release (v4.2 SLIM) 
http://www.retrorangepi.download/download/index2.html

closed 2018-08-01 by retr0rangepi