retr0rangepi/RetrOrangePi #21: Support for XBOX controllers

boboche, 2018-03-22:
Xbox one controllers, they are rumbing when plugued with USB cable, so enough current, but are not picked up by any drivers.

retr0rangepi, 2018-03-22:
check this: http://orangepi.club/showthread.php?tid=2287&pid=3923#pid3923

boboche, 2018-03-22:
Great! will go try it out, but why not making it supported out of the box, smoother first time experience (because so far, this is the only retro project that got me saying "Woah! Nice!" when applying power to the board (retro loader + music while installing - Awesome...Music at the intro stage, Sick.... xbox controller not working... reminds me of all of the other retro projects (minus one that supported them out of the box (recalbox if I remember correctly).

Thanks for the lightning quick reply, and good work!

Cheers,

B.

retr0rangepi, 2018-03-22:
because Xbox One is not properly supported with the legacy kernel, and since we dont have XOne to test, never wanted to blacklist xpad and risk breaking xbox360 support :p

boboche, 2018-03-22:
Oh ok! I can offer basic QA on the Xbox One support if you would integrate but have no 360 controllers here.  

I seriously regret having bought that controller, more problems than good for retro gaming :)

retr0rangepi, 2018-03-22:
yeah, looks like it needs mainline kernel for full compatibility. Just let me know if it works for you and i'll try to make my x360 work, and then push it to the git

boboche, 2018-03-22:
Not working, I recall having the same problem with retropie and this controller, there were 2 revs of Xbox One controllers (firmware revisions reporting different device ID, probably this is related.

https://github.com/RetroPie/RetroPie-Setup/issues/1277

retr0rangepi, 2018-03-22:
Are you trying it wired, no? Unfortunately we cannot reproduce, maybe if you pm the lucky user, he can guide you ... let us know if you get something

boboche, 2018-03-22:
Yeah wired.  Reporting vendorID=045e idProduct=02dd.  Tried to apply the steps in the previous message but its linux kernel based so obviously didn't get really far.

Installing and compiling the xpad driver should have worked, can't beleive they wouldn't support this device way over a year later.  Oh well :(  Will revert to good ol xarcade to continue my journey and fight with this when I get more time :)

Thanks

xgpt, 2018-04-13:
Bump? I'm not sure what to do here, I just bought two Xbox360 controllers thinking...they'd work easily...but now...one of them works and the other doesn't. Although the other one *does* work on my ubuntu laptop...it's not working on my brand new Orange Pi PC...HELP! please!

retr0rangepi, 2018-04-13:
Xbox360 should work, both must be mapped in EmulationStation

retr0rangepi, 2018-04-13:
Or maybe they're one of of those that charges from USB?

Liontek1985, 2018-11-20:
I have tested wired xbox360, wired xbox one, wireless Xbox360! you can use and mix it

1.) Install ROPI (without plugged controller!, only keyboard)
2.) Install xboxdrv (retropie menu)
3.) edit int the folder "/etc/modprobe.d/" the blacklist config and add "blacklist xpad" without ""
4.) edit the "/etc/rc.local"

Add your controllers manualy
id = wired 
wid= wireless
----------------------------------------------
xboxdrv --trigger-as-button --id 0 --led 2 --deadzone 4000 --silent &sleep 1
xboxdrv --trigger-as-button --id 1 --led 3 --deadzone 4000 --silent &sleep 1
xboxdrv --trigger-as-button --id 2 --led 4 --deadzone 4000 --silent &sleep 1
xboxdrv --trigger-as-button --id 3 --led 5 --deadzone 4000 --silent &sleep 1
----------------------------------------------

xboxdrv --trigger-as-button --id 0 --led 2 --deadzone 4000 --silent &sleep 1
**id** says its a wired controller
replace it with **wid** you says is a wireless controller

you can mix wired and wireless !!!

5.) restart retropie
6.) shutdown retropie
7.) plug your controller & start retropie
8.) set your buttons
9.) finish play and lought ^^

All Controller Works with player LED! no blinking playerstatus!

You want for every Controller a own config or special stats look here
https://pingus.seul.org/~grumbel/xboxdrv/xboxdrv.html

**After the Install steps
-Xbox One wired controller must plugged everytime before start retropie when you mix wired and wireless!!!**

retr0rangepi, 2020-04-08:
Latest v4.3 with mainline kernel should support natively

closed 2020-04-08 by retr0rangepi