retr0rangepi/RetrOrangePi #23: Incorrect Downloadable Image Versions for Orange Pi Zero Plus 2 H3

MayMih, 2018-05-02:
On both servers there are images with RetroOrangePi **4.0** - _not_ 4.**1** as mentioned in download section!

P.S. Not sure where to post it here or on the forum (but I can't register there).

retr0rangepi, 2018-05-02:
It's 4.1 indeed. EmulationStation menu will mention 4.0 (sorry about that) because the binary was only updated afterwards. 
4.0 version only exists by manual update from 3.0.1. It has three partitions, while 4.1 has only two.

closed 2018-05-02 by retr0rangepi

MayMih, 2018-05-02:
Ok, thank you for explanation. But, nevertheless, it would be good if somewhere on the site there were a notice that the user must update sytem first befor trying to instal in on eMMC for example.

retr0rangepi, 2018-05-02:
eMMC install was fixed after v4.1 release, we are working on v4.2 update, so expect more info to go public soon, thanks