retr0rangepi/RetrOrangePi #25: Sound configuration file /etc/asound.conf is wrong

yam001, 2018-05-03:
It should be defined as follows (note the missing exclamation signs). I tested it by changing the default  from 0 to audio device 3. Now, the sound works outside of emulation station as well. The sound was not being routed to the mixer correctly.

pcm.!default {
        type plug
        slave.pcm "dmixer"
}

pcm.dmixer  {
        type dmix
        ipc_key 1024
        ipc_perm 0666
        slave {
                pcm "hw:3,0"
                period_time 0
                period_size 1024
                buffer_size 4096
                rate 44100
                format S32_LE
                channels 2
        }
        bindings {
                0 0
                1 1
        }
}

ctl.dmixer {
        type hw
        card 3
}

ctl.!default {
        type hw
        card 3
}

retr0rangepi, 2018-05-03:
AFAIK, there is no such thing as card 3. Run 'aplay -l' and you will see two cards: 0 (audiocodec) and 1 (sndhdmi) , my guess is that selecting 3 will make it fallback to whatever default the system finds. By the way, ~/.asoundrc  adds extra userwise settings . Btw, sound was working fine when image was flashed, so i'm closing this for now

closed 2018-05-03 by retr0rangepi

yam001, 2018-05-03:
I forgot to mention the reason for the 4 cards is:

0: analog audio
1: i2s DAC
2: hdmi
3: USB DAC

I connected all of them to my Zero :-) Now, they all work by changing the number in the config file, whereas before only analog works.

retr0rangepi, 2018-05-03:
OK, that explains, so default setting is correct . We added an USB option (default to hw:2,0) to our Audio/Video Switcher bashtool, but it's hard to standardize.  Some boards, like the Banana Pi M2+ have only one audio device