retr0rangepi/RetrOrangePi #29: WiFi Not working on Banana Pi M2+ Zero

withgallantry, 2018-08-14:
Wifi isn't working on the Banana Pi M2 Zero with the RetrOrangePi 4.2 SLIM Image. 
Setting the wifi manually in /etc/network/interfaces using the below does not work. Making it impossible to install things such as GPIO controls. 
```
source /etc/network/interfaces.d/*
    allow-hotplug wlan0
    auto wlan0
    iface wlan0 inet dhcp
    wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
```

retr0rangepi, 2018-08-14:
Interfaces file is ignored by NetworkManager, looks like a latest Armbian update broke DNS resolving, still troubleshooting

retr0rangepi, 2018-09-04:
try this:
wget http://www.retrorangepi.org/interfaces
sudo cp interfaces /etc/network
sudo service network-manager restart
sudo ifconfig eth0 down

closed 2018-09-10 by retr0rangepi