retr0rangepi/RetrOrangePi #38: SDL2 configuration?

midwan, 2019-01-13:
I tried testing the current Amiberry, and it fails to open a screen on startup.

Specifically, we are thrown back to the console with this error:
`Unable to create window Could not get EGL display` which comes from SDL2 while trying to open the GUI window.

What configuration flags are you using for SDL2?
Is there a KMSDRM video back-end available for the platform?

retr0rangepi, 2019-01-13:
which kernel are you testing on? GPU support in Mainline seems to be missing some features

retr0rangepi, 2019-01-13:
on topic, my SDL2 flags are (--enable-video-mali --enable-video-opengles --disable-video-opengl --disable-video-vulkan --disable-video-x11) . 
i just returned from a short vacation, resuming beta testing the new Mali modules/libs. I get Amiberry to run (GUI), but it blanks the screen

KMS/DRM is apparently present, but i didnt test it too much:

pi@orangepipcplus:~$ sudo modinfo sun4i-drm
filename:       /lib/modules/4.14.83-sunxi/kernel/drivers/gpu/drm/sun4i/sun4i-drm.ko
license:        GPL
description:    Allwinner A10 Display Engine DRM/KMS Driver
author:         Maxime Ripard <maxime.ripard@free-electrons.com>
author:         Boris Brezillon <boris.brezillon@free-electrons.com>
alias:          of:N*T*Callwinner,sun8i-v3s-display-engineC*
alias:          of:N*T*Callwinner,sun8i-v3s-display-engine
alias:          of:N*T*Callwinner,sun8i-h3-display-engineC*
alias:          of:N*T*Callwinner,sun8i-h3-display-engine
alias:          of:N*T*Callwinner,sun8i-a33-display-engineC*
alias:          of:N*T*Callwinner,sun8i-a33-display-engine
alias:          of:N*T*Callwinner,sun6i-a31s-display-engineC*
alias:          of:N*T*Callwinner,sun6i-a31s-display-engine
alias:          of:N*T*Callwinner,sun6i-a31-display-engineC*
alias:          of:N*T*Callwinner,sun6i-a31-display-engine
alias:          of:N*T*Callwinner,sun5i-a13-display-engineC*
alias:          of:N*T*Callwinner,sun5i-a13-display-engine
alias:          of:N*T*Callwinner,sun5i-a10s-display-engineC*
alias:          of:N*T*Callwinner,sun5i-a10s-display-engine
depends:
intree:         Y
name:           sun4i_drm
vermagic:       4.14.83-sunxi SMP mod_unload ARMv7 thumb2 p2v8

midwan, 2019-01-13:
@retr0rangepi 
Sorry, I forgot to mention platform details:
I'm testing on an OrangePi PC, using `RetrOrangePi_4.2_FULL_Armbian_5.59_Orangepipc_Debian_jessie_default_3.4.113_desktop`

retr0rangepi, 2019-01-13:
v4.2 is still on legacy kernel so it should work fine. Try forcing framebuffer libs with LD_LIBRARY_PATH=/usr/local/lib

retr0rangepi, 2019-01-13:
latest version needs tidying, there are probably SDL2 headers in the wrong place (default repo vs retropie fork with mali support )
try this binary: http://retrorangepi.download/amiberry.tar.gz
anyway, i'm redownloading v4.2 to reproduce the behaviour

midwan, 2019-01-13:
`sdl2-config --cflags --libs` reports:
`-I/usr/local/include/SDL2 -D_REENTRANT`
`-L/usr/local/lib -Wl,-rpath,/usr/local/lib -Wl,--enable-new-dtags -lSDL2`

It seems correct to me, if those are the locations of the libs and headers indeed.
There's no need to try an older binary, as the whole point of my testing was to see how well the latest version runs before releasing a new stable :)

retr0rangepi, 2019-01-13:
Cool! I'll try my luck and let you know, booting a fresh image now
btw, had this report from one of our users (a RetroStone owner)

Hi, did a lot of testing last night and it really great to have Amiberry working. Most games run great but some slow down. Games like Top Gear 2 (AGA games) or Creatures run slow. With changing some settings i can get it to run a bit faster. Sometimes it needs a couple of times to start a game. Only thing that is a bit annoying is the screenresolution. Some games are not centered but you know. That’s difficult for Amiga. Maybe in the future Midwan comes up with a trick to auto adjust the screen. Thanks again for all your work!

midwan, 2019-01-13:
@retr0rangepi 
If you use the WHDLoad booter feature, then games use settings adjusted per title, using a local XML database. Things like screen size and centering are part of that. ;)

midwan, 2019-01-14:
@retr0rangepi 
Thanks, I'll test that on my side later and report back.
If it works this way, will you take care of updating the distro parts so that people won't run into this?

retr0rangepi, 2019-01-14:
Will do, let me know your results anyway

midwan, 2019-01-14:
@retr0rangepi 
For me, it wasn't enough to just remove the LD_LIBRARY_PATH line. I had to replace the value with /usr/local/lib in order to make it work.

Once I did that, it started up.

midwan, 2019-01-14:
@retr0rangepi 
Also, the `+Start Amiberry.sh` file has incorrect contents:

`cd /opt/retropie/emulators/uae4arm; LD_LIBRARY_PATH=/usr/lib/GLSHIM:/usr/lib startx /opt/retropie/emulators/amiberry/amiberry.sh`

retr0rangepi, 2019-01-14:
Yes, in default retropie , Amiberry script inherits UAE4ARM configs. As it needs X11 in ROPi, , i'll have to separate the emulator entries, or maybe get rid of uae4arm

retr0rangepi, 2019-01-18:
here it goes:

https://github.com/retr0rangepi/RetroPie-Setup/commit/6bdb1aa33a45b1465f86c578d72c168158dccac1

closed 2019-01-18 by retr0rangepi