retr0rangepi/RetrOrangePi #42: Retroarch won't launch over retropie menu

Liontek1985, 2019-02-18:
Retroarch won't launch over retropie menu

Hello Guys,

I have updated retropie script and then retroarch, now i can´t launch retroarch over retropiemenu
after reflashing the sd card i have make the same stepps and have the self effect...

over emulationstation i can launching games with lr-emulators without problems, only over retropiemenu i cant start it. I see no error only black screen and fallback to retropiemenu

i have tested to change the version to 1.7.5 and here the same problem, i think is a problem from the update script.

best regards fredy

retr0rangepi, 2019-02-18:
cannot reproduce this with v1.7.6, Retroarch runs fine standalone from ES. 
What are your steps to update Retroarch?

Liontek1985, 2019-02-18:
Retropie setup, main packages, retroarch, update

Liontek1985, 2019-03-15:
error fixed, is in retropie-setup -> closed

how you can fix? update the joy2key.py
i have manualy fork from original retropie-setup for pi3, manual edits in some files necessary

closed 2019-03-15 by Liontek1985

asiapool, 2021-11-01:
pi4 4.7.1  cant start retroarch confirmed