retr0rangepi/RetrOrangePi #44: Update Emulationstation over Retropie-Setup

Liontek1985, 2019-03-07:
Hello Guys,

When iam update Emulationstation or install Emulationstation-dev i get a older version from v4.1 and not the actual 4.2
With all old menu entries....

I am not alone with the problem, i make only the issue here
http://orangepi.club/showthread.php?tid=3087

https://github.com/retr0rangepi/EmulationStation/blob/master/es-app/src/guis/GuiMenu.cpp
GuiMenu::GuiMenu(Window* window) : GuiComponent(window), mMenu(window, "ROPi 4.1 MENU"), mVersion(window)

please update the source and fix the problem thx

retr0rangepi, 2019-03-07:
this should fix 
https://github.com/retr0rangepi/RetroPie-Setup/commit/061c06b1b5def0524c6e8fa7b587a746911d2714

Liontek1985, 2019-03-07:
very fast fix thx, i write it in the board

Liontek1985, 2019-03-09:
now works the update and i have follow error 
lvl0: VolumeControl ::init() failed to find mixer elements

sound works... always the volume error and in sound settings is volume 0% can change it to 100 after reopen the menu 0% and volume error

retr0rangepi, 2019-03-09:
ok, will take a look at this closer to release

Liontek1985, 2019-03-15:
i have change in es_settings the AudioDevice from Master to lineout now is the error solved
now not the best solution, better is an option to change the hdmi volume

retr0rangepi, 2019-03-15:
Probably will have to branch ES because of mainline and other settings if we  move completely to pulseaudio