retr0rangepi/RetrOrangePi #45: Problems with Translation to Portuguese PTBR 4.2

guinaldi, 2019-03-12:
Hi!

When I try to change language on 4.2 Slim Full to portuguese pt-br, shows me an error and i can't boot anymore. The error is: /opt/retropie/supplementary/emulationstation/emulationstation: error while loading shared libraries: libbost_system.so.1.55.0: cannot open shared object file: No such file or directory
/opt/retropie/configs/all/autostart.sh: line 11: 2490 Terminated            python /home/pi/RetrOrangePi/Bacground_Music/bgmusic.py

Thanks

Liontek1985, 2019-03-15:
here the same with change the language to german, i think this option is not stable

retr0rangepi, 2019-03-15:
Translations feature is not completed for v4.2 , so the script still fetches an old Emulationstation binary (with libboost dependencies).
It’s likely to come in mainline kernel with v4.3, run retropie-setup from command line , update to the latest version and install ES again (you can check temp folder, maybe the stock binary is still there )