retr0rangepi/RetrOrangePi #46: Can't update packages on retrorange Pi 4.2 full version

qu1que, 2019-03-29:
Hi!

In the retrostone console, which has retrorange pi v. 4.2 full version installed, from last week until today, when I try sudo apt update I've found the following errors:

`pi@RetrOrangePi:~$ sudo apt update
Obj http://apt.armbian.com jessie InRelease                                           
Obj http://security.debian.org jessie/updates InRelease                               
Des:1 http://apt.armbian.com jessie/main armhf Packages [453 kB]                      
Des:2 http://security.debian.org jessie/updates/main armhf Packages [794 kB]          
Des:3 http://security.debian.org jessie/updates/contrib armhf Packages [994 B]        
Des:4 http://apt.armbian.com jessie/jessie-utils armhf Packages [7.944 B]             
Des:5 http://security.debian.org jessie/updates/non-free armhf Packages [4.421 B]     
Des:6 http://apt.armbian.com jessie/jessie-desktop armhf Packages [22,6 kB]           
Ign http://httpredir.debian.org jessie InRelease                                      
Ign http://httpredir.debian.org jessie-updates InRelease                             
Ign http://httpredir.debian.org jessie-backports InRelease                   
Obj http://httpredir.debian.org jessie Release.gpg      
Ign http://httpredir.debian.org jessie-updates Release.gpg
Ign http://httpredir.debian.org jessie-backports Release.gpg
Obj http://httpredir.debian.org jessie Release        
Ign http://httpredir.debian.org jessie-updates Release
Ign http://httpredir.debian.org jessie-backports Release
Des:7 http://httpredir.debian.org jessie/main armhf Packages [8.898 kB]
Des:8 http://httpredir.debian.org jessie/contrib armhf Packages [44,3 kB]                                                                                                                                         
Des:9 http://httpredir.debian.org jessie/non-free armhf Packages [74,9 kB]                                                                                                                                        
Err http://httpredir.debian.org jessie-updates/main armhf Packages                                                                                                                                                
  
Err http://httpredir.debian.org jessie-updates/main armhf Packages                                                                                                                                                
  404  Not Found [IP: 151.101.120.204 80]
Err http://httpredir.debian.org jessie-updates/contrib armhf Packages                                                                                                                                             
  404  Not Found [IP: 151.101.120.204 80]
Err http://httpredir.debian.org jessie-updates/non-free armhf Packages                                                                                                                                            
  404  Not Found [IP: 151.101.120.204 80]
Err http://httpredir.debian.org jessie-backports/main armhf Packages                                                                                                                                              
  404  Not Found [IP: 151.101.120.204 80]
Err http://httpredir.debian.org jessie-backports/contrib armhf Packages                                                                                                                                           
  404  Not Found [IP: 151.101.120.204 80]
Err http://httpredir.debian.org jessie-backports/non-free armhf Packages                                                                                                                                          
  404  Not Found [IP: 151.101.120.204 80]
Descargados 10,3 MB en 42s (245 kB/s)                                                                                                                                                                             
W: Fallo al obtener http://httpredir.debian.org/debian/dists/jessie-updates/main/binary-armhf/Packages  404  Not Found [IP: 151.101.120.204 80]

W: Fallo al obtener http://httpredir.debian.org/debian/dists/jessie-updates/contrib/binary-armhf/Packages  404  Not Found [IP: 151.101.120.204 80]

W: Fallo al obtener http://httpredir.debian.org/debian/dists/jessie-updates/non-free/binary-armhf/Packages  404  Not Found [IP: 151.101.120.204 80]

W: Fallo al obtener http://httpredir.debian.org/debian/dists/jessie-backports/main/binary-armhf/Packages  404  Not Found [IP: 151.101.120.204 80]

W: Fallo al obtener http://httpredir.debian.org/debian/dists/jessie-backports/contrib/binary-armhf/Packages  404  Not Found [IP: 151.101.120.204 80]

W: Fallo al obtener http://httpredir.debian.org/debian/dists/jessie-backports/non-free/binary-armhf/Packages  404  Not Found [IP: 151.101.120.204 80]

E: No se han podido descargar algunos archivos de índice, se han omitido, o se han utilizado unos antiguos en su lugar.
`

I was considering to update from jessie to stretch changing the distro word on /etc/apt/sources.list, but I don't know if that will work.

Any idea on how to solve this?

HereticOne, 2019-04-01:
Note that i am not even able to connect to any websites.
is this https://github.com/retr0rangepi/RetrOrangePi/issues/47
definitely related?

Liontek1985, 2019-04-05:
look here http://orangepi.club/showthread.php?tid=3109

i have readed jessie debian is end of life! the update source dont work more

qu1que, 2019-04-05:
That is what I supposed, but the other jessie repositories works fine, only the jessie-backports and jessie-updates does not work

retr0rangepi, 2019-04-07:
From EmulationStation - Retropie - RetrOrangePi section, 
select 'Update RetrOrangePi' . It will fetch new repositories info.

Please also update RetroPie-Setup script

closed 2019-04-07 by retr0rangepi

HereticOne, 2019-04-09:
How do i do that if my device cannot resolve a DNS?

retr0rangepi, 2019-04-09:
There is not a perfect solution, as it's messed up in Debian Jessie. 

Things to try:

Depending on the board, it magically works after you remove an interface with "sudo ifconfig eth0 down" but you can also try try reconfiguring the resolvconf package with "sudo dpkg-reconfigure resolvconf" or editing network settings with "sudo nano /etc/network/interfaces" , remove the eth0 portion or "sudo nano /etc/NetworkManager/NetworkManager.conf", under [main] section , change to 'dns=dnsmasq' and/or under [ifupdown] section , change to 'managed=true'. 

after any change, run "sudo service network-manager restart"