retr0rangepi/RetrOrangePi #47: Can't connect or resolve domains not sure.

HereticOne, 2019-03-30:
Purely cannot get connected to update or access the internet on banana pi m2 zero.

retr0rangepi, 2019-03-31:
related to https://github.com/retr0rangepi/RetrOrangePi/issues/46

retr0rangepi, 2019-04-09:
reopening this as i have a bpi zero to test :p

retr0rangepi, 2019-04-19:
fresh v4.2 image
option RetrOrangePi Wifi from RetroPie - RetrOrangePi
Activate connection

wifi is up on wlan2 interface
SSH connection works

Using username "pi".
pi@192.168.0.20's password:
 ____      _         ___                             ____  _   _  _    ____
|  _ \ ___| |_ _ __ / _ \ _ __ __ _ _ __   __ _  ___|  _ \(_) | || |  |___ \
| |_) / _ \ __| '__| | | | '__/ _` | '_ \ / _` |/ _ \ |_) | | | || |_   __) |
|  _ <  __/ |_| |  | |_| | | | (_| | | | | (_| |  __/  __/| | |__   _| / __/
|_| \_\___|\__|_|   \___/|_|  \__,_|_| |_|\__, |\___|_|   |_|    |_|(_)_____|
                                          |___/
       █              █
  █     █            █     █
  █  ████████████████████  █
  ██████████████████████████
  ██████████████████████████
    ██████████████████████
     ████████████████████
        █             █
       █               █

Welcome to ARMBIAN Debian GNU/Linux 8 (jessie) 3.4.113-sun8i
System load:   0.63 0.83 0.41   Up time:       5 min
Memory usage:  13 % of 494MB    Swap usage:    5 % of 247Mb     IP:            192.168.0.20
CPU temp:      47°C
Usage of /:    12% of 29G

[ General system configuration (beta): armbian-config ]

Last login: Tue Sep  4 02:32:00 2018

pi@RetrOrangePi:~$ wget http://www.retrorangepi.org/es_systems.cfg
--2019-04-20 01:44:16--  http://www.retrorangepi.org/es_systems.cfg
Resolving www.retrorangepi.org (www.retrorangepi.org)... 173.236.154.62
Connecting to www.retrorangepi.org (www.retrorangepi.org)|173.236.154.62|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 25344 (25K) [text/plain]
Saving to: ‘es_systems.cfg’

es_systems.cfg                            100%[=======================================================================================>]  24.75K   150KB/s   in 0.2s

2019-04-20 01:44:17 (150 KB/s) - ‘es_systems.cfg’ saved [25344/25344]

pi@RetrOrangePi:~$

Issue not reproducible.

closed 2019-04-19 by retr0rangepi

retr0rangepi, 2019-04-20:
more workarounds:

pi@RetrOrangePi:~$ nano /opt/retropie/configs/all/autostart.sh

watch "sudo ifconfig eth0 down" &

pi@RetrOrangePi:~$ sudo nano /etc/network/interfaces.d/eth0.conf

include the following line:

#!/bin/bash
watch -n 1 'sudo ifconfig eth0 down' &>/dev/null &

btw, BPi Zero wifi is really unreliable. I got myself an usb ethernet dongle

HereticOne, 2019-05-04:
what image are you using the full or slim?
Tried both and on both i cannot resolve anything using both wifi and ethernet.
but i am connected and can see the device on the local network.