retr0rangepi/RetrOrangePi #48: Can't install packages due to libpng-dev causing errors

furyrulez, 2019-04-08:
[image]
I updated both Retrorangepi and the Retropie-setup script... but I still get this message when I try to install dosbox-sdl2, lr-caprice, eduke32... Is this normal? (I updated Mame2003 from source, installed Tyrquake, and some other stuff with no problem, btw).

retr0rangepi, 2019-04-09:
fixed 
https://github.com/retr0rangepi/RetroPie-Setup/commit/6966ff55fe4cfa1a9b9714df22a792c71d8ec8f9

closed 2019-04-09 by retr0rangepi

furyrulez, 2019-04-09:
I updated the retropie-setup script again, then I tried to install "eduke32" (Duke Nukem 3D port), and it still can't be installed. It stops during the process, and on the screen it writes "failed building sdlmusic.o from source/duke3d/src/sdlmusic.cpp" before exiting.
Then I tried "super flappy birds", and it can't be installed. 
Plus: Not sure if related to the update, but today I got 2 errors while browsing systems in emulationstations with the background music ON, they said something about an error with the audio mixer, audio card... I can't remember exactly.

retr0rangepi, 2019-04-09:
Open issues individually. Include correspondent info from ~/RetroPie-Setup/logs