retr0rangepi/RetrOrangePi #52: sunxi_cedar_mod g2d_23

Konstantin22222, 2019-07-06:
I already fix g2d_23 with installing xserver-xorg-video-fbturbo and creating then removing config but i dont know how to fix sunxi_cedar_mod. This modules load by kernel. Maybe need to recompile kernel without this modules?

retr0rangepi, 2019-07-06:
the sunxi_cedar_mod absence error doesn't seem to affect usability and performance. We never tested blacklisting it or recompiling the kernel. Since we're moving to mainline kernel, only legacy severe bugs should be fixed.

Konstantin22222, 2019-07-06:
Thus module need to be installed because with it people can watch video on orange pi pc without lagging (i mean hard for gpu video like 1080 .mkv file)

Konstantin22222, 2019-07-06:
And will be kodi update? Some plugins need 18 version

retr0rangepi, 2019-07-06:
hardware acceleration in legacy is provided by VDPAU and MPV as external player. It works like this in Kodi. 
Yes, Kodi 18 will be included in the upcoming release (mainline kernel)

Konstantin22222, 2019-07-06:
It's cool! And last thing, scripts from desktop to bootup armbian, emulationstation and etc. are not working after making external usb for roms. I use method for retropie, just make folder retropie-mount and enable external usb rom option in retropi-setup. Maybe i do something wrong?

retr0rangepi, 2019-07-06:
those shortcuts are not suitable for retropie-mount, as we use another approach for using external usb. This will be sorted out for the next version as well

Konstantin22222, 2019-07-06:
Okay, thanks! You doing a great job, thanks for the system!