retr0rangepi/RetrOrangePi #54: custombluez OpiPC2 wont compile

thedotsteve, 2019-11-10:
appears to be compiling as ArmV7 (32 bit) and needs to be ArmV8 (64 bit)

thedotsteve, 2019-11-10:
[rps_2019-11-10_162015.log](https://github.com/retr0rangepi/RetrOrangePi/files/3828705/rps_2019-11-10_162015.log)

thedotsteve, 2019-11-10:
after several attempts to compile bluez from source i suspect it needs to be crosscompiled on an X86 based machine and installed as a .deb package
its dep versions are not released on arm

thedotsteve, 2019-11-13:
i think i may have found the issue

PKG_CONFIG_PATH isnt working correctly
only default paths are checked

suspect this is effecting most compiler functions

retr0rangepi, 2019-11-13:
Important: for several reasons, the PC2 image was set to compile against 32bit. 
A pure 64bit image will be compiled soon , if most emulators work

edneyhelene, 2019-11-13:
> Important: for several reasons, the PC2 image was set to compile against 32bit.
> A pure 64bit image will be compiled soon , if most emulators work

 May i know please where the build file is located? i'm willing to find out which version of reicast are you guys using because its super good!

thedotsteve, 2019-11-14:
Wish I'd known that earlier

retr0rangepi, 2019-11-14:
It was noted on the Facebook thread that everything was 32bit but PSP and Reicast. Apparently not clear enough, apologies.

retr0rangepi, 2020-02-20:
fixed on beta 64bits image

closed 2020-02-20 by retr0rangepi