retr0rangepi/RetrOrangePi #57: Kodi 18.4 Leia

macchese-eng, 2019-11-17:
Is it possible to have 18.4 Leia as version of Kodi? With the 19 Alpha it is impossible to load the addons, and I was interested in how the tvheadend works Thanks and congratulations for the work done

retr0rangepi, 2019-11-17:
addons work on kodi 19 as well.

Instructions:

Quit Emulationstation (F4)
cd
wget http://www.retrorangepi.org/addons.tar.gz
sudo tar -zxvf addons.tar.gz -C /usr/share/kodi/addons/ 
(case sensitive)
Includes: peripheral.joystick (gamepad input support), pvr.iptvsimple and pvr.hts (headend)

For IPTV , you'll have to go to settings - add-ons - PVR clients -  Enable

[image]

closed 2019-11-17 by retr0rangepi

macchese-eng, 2019-11-17:
I have only pvr-iptv...no Tvheadend Thanks

retr0rangepi, 2019-11-17:
you mentioned tvheadend , anyway iptvsimple should work for you

macchese-eng, 2019-11-17:
Thanks anyway. But as pvr I use tvheadend for a TV that has no possibility of terrestrial antenna. I will wait for new updates. Thanks again for the job done