retr0rangepi/RetrOrangePi #59: Tritium H3 build for V4.3

keiem, 2019-11-23:
Hello

are there any plans to build rtrorangepi 4.3 for the tritium H3.  Or can I just download any H3 version and change the DTB?

retr0rangepi, 2019-11-27:
Please test: http://retrorangepi.download/RetrOrangePi_v4.3_Armbian_19.11.3_Tritium-h3_bionic_current_5.3.13_desktop.img.7z

keiem, 2019-11-28:
Hello,

it doesn't boot
"found U-boot script /boot/boot.scr
Executing script at 43100000
Wrong image format for "source" command
SCRIPT FAILURE: continuing
card did not respond to voltage select

Device 0: unknown device_sun8i_emmc_eth_init: Timeout
missing environment variable:pxeuuid
missing environment variable: bootfile
..."

retr0rangepi, 2019-11-28:
New image is up, same link. Please try again and post full log (picture)

keiem, 2019-12-04:
Thanks, the image boots and everything seems to work.  Although I didn't test some games.  I can tell eduke32 wouldn't install and I can't seem to find super mario war in the optional packages.  Any reason why it is removed?

retr0rangepi, 2019-12-04:
Thanks for the info. Please open new issues to fix eduke32 and smw . The latter is not compatible with the KMS mode (new standard for mainline kernel) - https://github.com/retr0rangepi/RetroPie-Setup/blob/8996e64725fc30cafad56d58e52cc75b61333b56/scriptmodules/ports/smw.sh#L16
Since image works, closing this one.

closed 2019-12-04 by retr0rangepi