retr0rangepi/RetrOrangePi #70: ROPI 4.3 support for OrangePi Plus 2

ribasrogerio, 2020-05-05:
Hi, I'm trying to run RetrOrangePi v4.3 on OrangePi Plus 2 with image for OrangePi Plus 2e.
Armbian does not offer a device tree file for the OPi Plus 2 so to make USB work.
If there a chance to run RetrOrangePi on this board?
I can post a logs if needed.
Thanks!

svirot, 2020-05-06:
Using the "OrangePi Plus 2e" release and adding `fdtfile=sun8i-h3-orangepi-plus.dtb` to /boot/armbianEnv.txt solved the issue for me.

grawi-sikorsky, 2020-06-15:
Just saw this issue, i'll try for sure and give feedback. But for now i have different problem. On SD card (img flashed by Etcher) there's no boot partition visible on Win7 :( Can't find solution for this.

retr0rangepi, 2020-06-15:
RetrOrangePi is a Linux OS with ext4 filesystem, not supported natively in Windows. Alternatives: use 3rd party apps like Paragon or Ext2Fsd or boot a Live CD/USB with any Linux OS