retr0rangepi/RetrOrangePi #74: Retrostone 2 - driver issue

maximilianvoss, 2020-06-11:
Hi,
it may happen during playing that the driver runs into a state where the left button is pressed permanently. After killing and restarting the process of 
```bash
/home/pi/RetrOrangePi/GPIO/drivers/tz_gpio_controller.py
```
it works perfectly fine again.
I have the feeling it might happen if a lot of buttons are pressed at the same time as it happens more often in challenging games and levels.

megashub, 2020-06-20:
+1 -- I've seen this symptom too.

maximilianvoss, 2020-06-21:
I investigated further, it is a static charge on the joystick's solder points. If the joystick part from the driver is deactivated it works as it won't be passed along.

megashub, 2020-06-21:
Ooo... discouraging news... Has the RS2 hardware team been informed? Sounds like a miss during manufacturing, right?

Though I suppose you could always choose which type of controller you want to use for any given game and explicitly disable the unused one... but that just sounds painful to do title by title.

maximilianvoss, 2020-06-23:
if you ran in the same issues, I did a fix to the driver https://github.com/maximilianvoss/retrostone2 which disables the joystick