retr0rangepi/RetrOrangePi #81: [Retrostone 2] Scrapper not working due to old version of EmulationStation & Problems upgrading EmulationStation

mferrem93, 2021-01-27:
There's an issue when we try to use the built-in scrapper from EmulationStation. The frontend itself doesn't say anything, but if you take a look at the logs they say the next lines:

**lv10: TheGamesDBRequest - Error parsinig XML**

It happens the same if you try to use ScreenScrapper instead. Searching on forums I discovered that this Issue is caused because TheGamesDB made changes on its API and EmulationStation needs to be upgraded. So, if we try to upgrade EmulationStation via RetroPie-Setup, it says that was not possible to build it succesfully due to some errors. If we take a look more deeply into the logs given on build time:

> ...
> -- Found SDL2: /usr/local/lib/libSDL2main.a;/usr/local/lib/libSDL2.so;-lpthread  
> -- Found CURL: /usr/lib/arm-linux-gnueabihf/libcurl.so (found version "7.58.0") 
> -- Checking for module 'libvlc>=1.0.0'
> --   Found libvlc, version 3.0.8
> -- VLC library found
> -- Found VLC: /usr/include/vlc  
> -- Could NOT find Rapidjson (missing: RAPIDJSON_INCLUDE_DIRS) 
> -- Found ALSA: /usr/lib/arm-linux-gnueabihf/libasound.so (found version "1.1.3") 
> CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
> Please set them or make sure they are set and tested correctly in the CMake files:
> RAPIDJSON_INCLUDE_DIRS (ADVANCED)
> ...

So, to solve this I had to install the package **rapidjson-dev** and then I could build succesfully and therefore upgrade EmulationStation.

After all of this I could use the scrapper with no problem.

I hope this helps.

mferrem93, 2021-01-27:
[rps_2021-01-27_111607.log](https://github.com/retr0rangepi/RetrOrangePi/files/5882159/rps_2021-01-27_111607.log)

retr0rangepi, 2021-01-27:
rapidjson-dev dependency was already set in the RetroPie-Setup fork [#branch v4.3](https://github.com/retr0rangepi/RetroPie-Setup/blob/5ae568232f5371ff6b39dc3793f7d0c0bae5f56e/scriptmodules/supplementary/emulationstation.sh#L131) . Looks like the local RetroPie-Setup folder in Retrostone2 and other was still pointing to an older branch. I believe a git pull or updating from its menu would fix this. Luckily, ES sources for the old branch were updated a week ago , that's why it worked (but you probably have an awkward ROPI 4.2 Reloaded in the main menu instead of 4.3). Apologies for the confusion. Anyway, both ES branches are updated now https://github.com/retr0rangepi/EmulationStation/commit/c1359f7a52a3fd426ee7074c6a7c62b2b94086bb

closed 2021-01-27 by retr0rangepi