retr0rangepi/RetrOrangePi #85: Is anyone maintaining this project somewhere?

ThallesP, 2022-11-14:
from the look of the commits, it seems that it isn't being actively maintained, did someone forked this project?

UncleRus, 2022-11-14:
http://retrorangepi.download/forum/showthread.php?tid=3759

wa1a, 2023-03-26:
Sadly http://retrorangepi.download  seems to be down, could anyone share some information about the project status?

ThallesP, 2023-03-26:
> Sadly http://retrorangepi.download seems to be down, could anyone share some information about the project status?

if you want to download the images, you can use this website https://www.retrorangepi.org/download/
i've used [rearm](https://rearm.it/) which is a different person maintaining it, works great with a few tweaks.

ThallesP, 2023-03-26:
i'll leave this open in case someone need this.